synthetos/TinyG #214: Move all motors at same velocity with combo G0 cmd

tiffyreeser, 2017-09-08:
Is it possible to configure TinyG in a way so that all motors will turn at the same rotational speed? 
e.g. given a cmd such as: 
g0 x10 y20 z30 
thx

cmcgrath5035, 2017-09-12:
How would you command the fourth motor in Gcode? As a rotational (A,B or C)?
Or would the 4th motor be a clone of X, Y or Z?

ril3y, 2017-09-12:
In short you would need to change your settings to reflect this.  If you
went x10 and y20 then your motors travel per mm would have to be 2x on x
than on y.  But I am not sure what you are after here?

Riley

CAReeser, 2017-11-29:
No longer an issue as tinyg computes best path and I just extended the computations to include the other 5 tinygs as one system. Now all 21 motors start at the same time and stop at the same time which is what nature does.  Thanks for your assistance.

closed 2017-11-29 by CAReeser