synthetos/TinyG #215: 2017 MacBook Pro unable to connect with USB-C adapter

Snortimer, 2017-10-11:
Howdy!

I've been scouring the interwebs, and I'm so far unable to find a solution. I recently replaced my 2012 MacBook Pro Retina with a 2017 MacBook Pro Touch Bar, and I can't connect to the TinyG (using Serial Port JSON Server) anymore.

DanalEstes, 2017-10-11:
Can you connect with a terminal?  On a Mac, it should be something like:

   - Open an OS X terminal session (window)
   - Find the right TTY device via an 'ls' command similar to:
   -
   - ls /dev/cu.*

   - Then use the mac built in program 'screen', and the device you found
   above, and the tinyg default baud rate of 115,200
   -
   - screen /dev/cu.usbserial 115200

   -
   - While in the screen app, you should be able to type commands, and get
   responses, as found in the tinyg documentation.
   - To quit the screen app, type CTRL-A, then CTRL-\.

ril3y, 2017-10-11:
What adapter are you using?  I have the same MacBook and it works for me.  However it is buggy.  Is this a v8 or a g2based board?  I have used a USB2 hub too to get it to work.  Crappy but not really something we can deal with.  

From: Joel Hall
Sent: Wednesday, October 11, 2017 6:35 PM
To: synthetos/TinyG
Cc: Subscribed
Subject: [synthetos/TinyG] 2017 MacBook Pro unable to connect with USB-Cadapter (#215)

Howdy!
I've been scouring the interwebs, and I'm so far unable to find a solution. I recently replaced my 2012 MacBook Pro Retina with a 2017 MacBook Pro Touch Bar, and I can't connect to the TinyG (using Serial Port JSON Server) anymore.

Snortimer, 2017-10-13:
Thank you both for your responses. I'll be testing these this afternoon and I'll get back to you with the results.

Snortimer, 2017-10-17:
Ok, so it took an extra couple days for me to get out to the garage and test things.

I was using Apple's Thunderbolt 3 to USB 3.1 adapter. Since its at the office right now, I just tested again with a cheap $10 Thunderbolt 3 to USB 3.0 adapter. No joy. Even after installing the driver again.

I tried to connect using Terminal, but it doesn't list the right ports. I tried with SPJS, as well, and it didn't do any better (obviously... why would it?).

So at this point, I'm still where I started. No connection. I'll see about getting a USB 2 or 3 hub if I can, and see if that works (I've got one laying around and I should be able to connect it with the Thunderbolt 3 adapter). I wonder if there's some issue with Thunderbolt 3. My old Mac had USB 3.0, so I know I was able to connect over that.

Thank you again for your help. Any other ideas?

DanalEstes, 2017-10-17:
Until the serial port appears, there is very little that Synthetos/TinyG
can do to help.

Snortimer, 2017-10-17:
Yeah, I figured as much. I'll have to keep bashing my head against it for a while. Or maybe install my Raspberry Pi.

giseburt, 2017-10-17:
I just happened to get a new MacBook Pro with the same configuration: 4
usb-c ports. I’ll test after work tonight and let you know what I find.

-Rob

Snortimer, 2017-10-17:
The new one has Thunderbolt 3 ports, so that might be part of the issue.

giseburt, 2017-10-17:
Thunderbolt 3 and USB-C are *almost* synonymous. Mostly. See
https://support.apple.com/en-us/HT207256

The part that they don’t explain is that USB-C doesn’t always carry
Thunderbolt 3 signals (it does on these MacBook Pros, but not on many
Chromebooks and other computers), but Thunderbolt 3 *does* always use a
UBC-C connector and carry USB 3+ signals.

So, a USB-C to USB Adapter like this one (or a cheaper one that you can
likely pick up at a local Target or WalMart) should do:
https://www.apple.com/shop/product/MJ1M2AM/A/usb-c-to-usb-adapter

-Rob

HoardOfGnomes, 2017-10-19:
We are trying to flash a Printrbot G2 v3c board using a 2017 MacBook Pro Touch Bar with a USB-C digital a/v multi-port adapter. We can't get it to connect with the firmware updater. It gives us a "extra argument found" error.

Snortimer, 2017-12-07:
@giseburt, that's exactly the adapter I tried using first, and it didn't work.

ril3y, 2017-12-08:
I have the same mac.  I use a hyper drive and even then I have to use a usb
hub at times to get it to work.  Its really hit or miss.  Try using a usb
hub with whatever converter you are using now.

Riley

BrettDikeman, 2018-01-25:
@Snortimer, what do you see in System Report under USB? Does the TinyG appear there?

GunnarAK, 2018-07-28:
Got the 2 port Macbook Pro (the one without touchbar) and I can't get it to work as well. 
The command `ls /dev/tty.*` only shows the bluetooth channel, which is unusable although I have a bluetooth module attached to the device (with correct baudrate).

I have the Satechi USB-C dongle with 1 HDMI, 2 USB-A and 1 USB-C port.

Hope someone can help us out!

imashankar, 2018-12-07:
Sorry to hit on a old thread. I bought a Mac book pro 2018 and facing same issue. Did anyone get it to a resolution?

GunnarAK, 2018-12-07:
Last time I checked I was able to connect via the USB hub I have. Solution: I'm not 100% sure it was because of updating to Mojave, but you could give it a try I guess. It's been a while

yshrini, 2018-12-21:
I am also facing same issue on my MacBook Pro (13-inch, 2017, Two Thunderbolt 3 ports)

giseburt, 2018-12-21:
Do you have the FTDI drivers installed?  OS X (I believe since Mavericks)
comes with some drivers but they don’t appear to expose the specific model
of chip that’s on the TinyG, but they load for that chip anyway. In short,
it appears like it’s going to work (shows up in the System Report under
USB) but doesn’t actually work.

You likely want one from here: http://www.ftdichip.com/Drivers/VCP.htm

I’ve been using this on my MacBook that only has USB-C ports with only a
simple adapter.

But you need those drivers.

-Rob

Snortimer, 2018-12-21:
I had the proper drivers with my previous Mac, and have installed them many times on my new one.

The 2016 model used USB-C ports with the USB protocol. The 2017 and later use Thunderbolt 3 over USB-C ports.

-Joel Hall

imashankar, 2018-12-30:
> Sorry to hit on an old thread. I bought a Mac book pro 2018 and facing same issue. Did anyone get it to a resolution?

I just did a hard reset on type-C ports. Also received a replacement adaptor from Amazon. Not sure what helped, but its working fine for me. Thanks a lot for all the help. 👍

maxbarbul, 2019-01-23:
@imashankar, thanks for the details.
I had the same issue and I fixed it by [resetting SMC](https://support.apple.com/en-us/HT201295)

closed 2019-07-29 by ril3y