synthetos/TinyG #217: Strange Behavior / Connection of Stepper Driver

DoktorKrupp, 2017-12-31:
Hello

I am new to CNC and to TinyG.
I have some problems with the board i guess.

I wired all 3 motors (Nema 23) into a separate driver (DM542T) which came with the motors. I soldered the cables from the tinyg to the driver. From the driver i wired it into the motors.
I have absolutely no problem with X Axis.
Its moving perfectly.

Strange Thing Number 1:
I have problems with my Y Axis. It only moves when i map it to two Axis (doesnt matter which one 2+3 or 2+4).
Z-Axis is not moving at all.
Is this normal? If yes: why?

Strange Thing Number 2:
When i connect the Z-Axis into the board (without driver) to motor 3, its working. On 4 its not working properly (only moving in one direction).
But Y is still not moving. Only if i map it to 2+4. And also here: only in one direction.
Is this normal that Y and Z is only moving to one direction?

I really would like to use the drivers because of the high amp of the motors.

I also tried it with the recommended Firmware (440.20) and the Experimental (442.04) plus the 449.01. All with the same result

Thanks a lot

BR

Manuel

DoktorKrupp, 2017-12-31:
Does it has something to do with the 5V input?

cmcgrath5035, 2018-06-28:
You are soldering to the driver input headers, correct?
Levels there are 3V logic, may be marginal drive for a 5V rated input on DM542T

How did you connect the ENA interface? tinyG is enable-active-low, DM542T is enable-active-high.
You should leave ENA not connected (default=enabled, per data sheet) for testing.

When using DM542t, are the tinyG microstep settings consistent with the manual switch settings on the DM542T?

ril3y, 2021-04-09:
Closed due to inactivity

closed 2021-04-09 by ril3y