synthetos/TinyG #219: chaining 2 tinyG boards

brettdaman, 2018-01-12:
I read and googled quiet a bit about this topic but could not find a working solution.
I know I can use 2 usb ports and connect them parallel but I would prefer it with a serial link between a master and a slave if possible. I read that in the past the timings would differ a little but for my usecase all the motors for board 1 are independent for board 2 so I dont care that they dont work in perfect sync. I just need to send motor commands, go to that position.

I started by connecting the miso,mosi,sck and ground.
Tried to map motor 5 to b but motor 5 throws an error if i write the settings 
So next i tried to configure motor 1 on the slave board to motor 5 and kept the settings on the master board as if there where only 4 motors. Didnt work either.

Can someone point me in the direction how to get the chainlink of tinyG working?
Can i flash new configuration over that link? 

Or is it just a lost cause?

johnlauer, 2018-01-12:
Not sure if this could work for your use case, but you could send data from ChiliPeppr. If your Gcode is XYZABC axes, you could remap the ABC to XYZ on the 2nd TinyG via a macro in ChiliPeppr. Or, perhaps if you’re scripting some kind of device you could just send via SPJS in parallel.

brettdaman, 2018-02-08:
I tried different setups but in the end I used 2 usb connections and solved it by abstracting my application to control them independently.

closed 2018-02-08 by brettdaman