synthetos/TinyG #220: Irradic Movement while cutting

CW3Davis, 2018-02-15:
I have the latest version of the TinyG controller. Sometimes I will be cutting and the zero position will shift by a half inch. This is during a cut when the machine is moving to a new location and continues to cut the file overlapping it. It doesn't do this all the time just once in a while, but I can't trust the controller to actually control the machine correctly to produce anything. 

I will not say I am an expert at setting this up. Most of what I have done is from looking at what others have their machines set to. I have 8mm leadscrews 2mm pitch on all axis. I have Nema 23 steppers 178oz motors. 24VDC power supply.

cmcgrath5035, 2018-06-28:
Does your zero move by 0.500", or does it wander by about a half inch?
If wander, do you have $_pm=2 on all motors?
$_pm=2 will hold other axes in place while one might be moving.
Are your motor to leadscrew connections tight? A loose allen screw can cause this as well.

ril3y, 2019-07-29:
closing due to inactivity

closed 2019-07-29 by ril3y