synthetos/TinyG #221: Post processor

nesquik011, 2018-02-20:
Hello i need 5 axis post processor please

nesquik011, 2018-02-20:
With auto tool changer

justinclift, 2018-03-18:
Hmmm, this might not be the right place to ask. :wink:

Generally a post processor is specific to both a software package (eg Fusion 360, Rhino, NX, etc), and to the machine it needs to generate G-Code output for.  eg a "ShopBot post processor for Fusion360", "ShoptBot post processor for Rhino", (etc).

Which software package do you need it for, and what is the output machine too? :smile:

nesquik011, 2018-03-18:
Fusion 360 , 5 axis  XYZ A and B  , and who makes them ? we need more post processor or a program create them to us

justinclift, 2018-03-18:
For Fusion 360, Autodesk seems to supply various standard ones, but there's also mention of them being willing to supply manuals and reference info to people wanting to make their own.

There is a list of the post processors that Fusion 360 already has here, with TinyG already on the list:

&nbsp; &nbsp; http://cam.autodesk.com/posts/

Reading through the TinyG post processor file now (for Fusion 360, they're written in javascript), it seems to be capable of doing 6 axis stuff.  There are variables and code things in it for x, y, z, a, b, c axis.  So you might be in luck already. :smile:

This thread seems useful too, as it explains where to start when thinking about making your own post or customising and existing one:

* https://forums.autodesk.com/t5/hsm-post-processor-forum/what-are-post-processors-and-how-do-i-get-them/td-p/5032332

Does that help? :smile:

nesquik011, 2018-03-18:
function onOpen() {
  if (properties.useRadius) {
    maximumCircularSweep = toRad(90); // avoid potential center calculation errors for CNC
  }
 
  if (!machineConfiguration.isMachineCoordinate(0)) {
    aOutput.disable();
  }
  if (!machineConfiguration.isMachineCoordinate(1)) {
    bOutput.disable();
  }
  if (!machineConfiguration.isMachineCoordinate(2)) {
    cOutput.disable();
  }
   
  if (!properties.separateWordsWithSpace) {
    setWordSeparator("");
  }

thats the tinyg post processor over there , it doesnt has 5 axis only 3

nesquik011, 2018-03-18:
http://cam.autodesk.com/posts/post.php?name=tinyg

nesquik011, 2018-03-18:
if it need to be edited to be 5 or 6 axis , may you do it and send it here or to my email please , dream_land_2@hotmail.com

justinclift, 2018-03-18:
Damn. Oh well. :frowning: 

I think you'll need to contact their CAM forum people then, and maybe either get them to send you the manuals and reference (to create the post yourself), or somehow engage them (or someone else) to create one for you. :smile:

It's not something I'll personally, as I no longer use Fusion 360.... and I *really* don't like javascript.  You'll need to contact the Autodesk people, as it mentions here:

&nbsp; &nbsp; https://forums.autodesk.com/t5/hsm-post-processor-forum/what-are-post-processors-and-how-do-i-get-them/td-p/5032332

nesquik011, 2018-03-18:
i did many time no answer , no one want to helps , so i thought about synthetos , tell fusion to make it or something , and why there is no program , has all the settings i pick whatever , then make the post for me so i can use it with whatever program i like and all companies unite in that step

justinclift, 2018-03-18:
It's a decent thought.  Making the (5 axis) post Fusion 360 is probably going to take many hours of effort though.  So, it'd probably cost several thousand $ of someone's time.

If you're willing to pay someone, then it shouldn't be too hard to find someone interested.  If it's something you need for free though, you'll probably need to take the time to create it yourself.  It *is* the kind of thing you can you, if you put enough time into it. :smile:

nesquik011, 2018-03-18:
i have been trying , i dont understand what should i put there , i swear i did try for months no hope  !  :(

justinclift, 2018-03-18:
How well do you program javascript?

Blueicaro, 2018-03-18:
<https://github.com/synthetos/TinyG/issues/221#issuecomment-373993743>, or mute
the thread

justinclift, 2018-03-18:
Ok.  You're going to need to teach yourself javascript first then.  Really.

It will take some time, but you can get it done if it's important enough to you. :smile:

justinclift, 2018-03-18:
@Blueicaro ???

nesquik011, 2018-03-18:
learn normal JS  ? of the web developing ?

justinclift, 2018-03-18:
@nesquik011 Yep.  Fusion 360 post processor files are written in javascript.  So, learn javascript, even though it's usually for web development.  When you understand javascript ok, you'll be able to understand the Fusion 360 post processor files and how to change them for what you need.

Make sense? :smile:

nesquik011, 2018-03-18:
ok another question can i use the post processor of pokect nc ? its 5 axis , and does every controller system has its own Gcode ? for example Tinyg and mach3 both are different in Gcode understanding even if i use cam on fusion same cam i mean ?

justinclift, 2018-03-18:
Good question. :smile:

With Pocket NC, I think they might use TinyG2 core themselves.  So their post processor might "just work" anyway. :smile:

justinclift, 2018-04-08:
This should help:

&nbsp; &nbsp; https://github.com/justinclift/Mill-Post-Processors-Maker-Machines/blob/TinyG-5th/tinyg.cps

It's the "standard" TinyG post processor that comes with Fusion360 (from [here](http://cam.autodesk.com/posts/), just now).  But it has a 4th and 5th axis added, using the steps from [here](https://forums.autodesk.com/t5/hsm-post-processor-forum/how-to-set-up-a-4-5-axis-machine-configuration/td-p/6488176).

You will probably want to change the rotation values [here](https://github.com/justinclift/Mill-Post-Processors-Maker-Machines/blob/5868915cccba46c6a4166e32f7130900a8d25b64/tinyg.cps#L151-L152) to suit your machine.

```
var aAxis = createAxis({coordinate:0, table:true, axis:[-1, 0, 0], range:[-30-0.0001,120+0.0001], preference:1});
var bAxis = createAxis({coordinate:1, table:true, axis:[0, 0, 1], range:[-999.999,999.999]});
```

The meaning of the values is explained here a bit.  They tell Fusion360 what the rotation limits are (and related things) for each axis:

* http://cam.autodesk.com/posts/reference/classPostProcessor.html#a8803b36be6893a81991049766abe0794

---

Note - there is [a line](https://github.com/justinclift/Mill-Post-Processors-Maker-Machines/blob/5868915cccba6c6a4166e32f7130900a8d25b64/tinyg.cps#L156) I am unsure about, in that code I cut-n-pasted for those 4th and 5th axes.

```
optimizeMachineAngles2(1); // map tip mode
```

I don't yet understand what `TCP` means in the Fusion360 CAM world, so I'm not sure if the value it sets (1) will mess things up for you or not.

Also, I have no idea how to add an Automatic Tool Changer.  It's not something the machine I'm building can do.  So I don't have an understanding of how Automatic Tool Changers work. (yet)

You can add it yourself though, if you work out the M and G-Code needed by your machine. :smile:

nesquik011, 2018-04-08:
Is there 5 axis post for any other milling cad cam softwares like power mill because fusion sucks with 5 axis

This should help:

    https://github.com/justinclift/Mill-Post-Processors-Maker-Machines/blob/TinyG-5th/tinyg.cps

  *   http://cam.autodesk.com/posts/reference/classPostProcessor.html#a8803b36be6893a81991049766abe0794

justinclift, 2018-04-08:
> ... sucks with 5 axis.

Yeah, no argument here.  But, you should probably work with what you can get.  At least when you're looking for favours. :grin:

justinclift, 2018-04-08:
@aldenhart This topic should probably be closed.  Doesn't seem like any use it leaving it to run?

aldenhart, 2018-04-08:
Agreed. Thanks for weighing in.

closed 2018-04-08 by aldenhart