synthetos/TinyG #223: Sending a reset command over serial

AND-Studs, 2018-03-16:
Hello!

I'm currently driving the tinyG with a custom app written in Java (its an Android App)

Thus far everything is working quite well! Today I tried to implement some behavior when a limit is hit and for the life of me can't get any message sent to the board to trigger a reset.

Pressing CTRL+X in coolterm triggers the reset immediately (whereas other commands require pressing Enter/Return) but in code nothing seems to work. I've tried strings of "^X", "^x", "\u0024" with and without "\r" (carriage return in java) with no success. Anything I might be missing? 

Thank you!

troisiemetype, 2018-03-17:
It's been too long since I've read the TinyG code, but my assumption is that ctrl+X may be sent by coolterm as flow control, which is "multiplexed" over USB, and that TinyG listens to it with its flow control pins (PC0 and PC1, CTS and RTS, resp.), not with the RX line, and generates an interrupt. And that in your case it comes on the RX line, and TinyG just discards it?

I don't know how you can verify it. The Synthetos team will give you a better answer...

DanalEstes, 2018-03-17:
You need to send the actual ASCII code that is generated by typing Ctl+X.
This is the ASCII "CAN" character, decimal 24, Hex 18, Octal 30.

So I believe you want something like:

     port.Send <https://strokescribe.com/en/serial-port-properties-and-methods.html#Send>(0x18);

Or similar.

AND-Studs, 2018-03-19:
Thanks for the replies! And a big thanks to @DanalEstes because you were correct sir!

I instantiated a byte array with one element b[0] = 0x18 then sent this array to the tinyG and voila! resets every time.

troisiemetype, 2018-03-19:
Nice to read You've find your way!!
I've realized after having written my comment above, that altough what I said was not false, it's the USB to serial chip job to do what I described... :/

ril3y, 2021-04-09:
Closed due to inactivity

closed 2021-04-09 by ril3y