synthetos/TinyG #224: Turn off LEDs

CAReeser, 2018-03-26:
Is there a command that will turn off or disable all LED's.  
Six TG's in a box emits a lot of light.

Thx
Cliff

troisiemetype, 2018-03-26:
Hi,

Unfortunately for you, since leds are connected between AOUT1 on TI stepper drivers, the only way to shut the leds would be to cut the trace near the leds.
That's true leds give a "let's go party" feel when a job is running!  :D

troisiemetype, 2018-03-26:
Hi,

Unfortunately for you, since leds are connected between AOUT1 on TI stepper drivers, the only way to shut the leds would be to cut the trace near the leds.
That's true leds give a "let's go party" feel when a job is running!

Have a look at the schematics, page 3 and 4 for more information.

DanalEstes, 2018-03-26:
Black tape.

CAReeser, 2018-03-26:
Yeah, I was thinking fingernail polish..
Thanks,
Cliff

From: Danal
Sent: Monday, March 26, 2018 2:11 PM
To: synthetos/TinyG
Cc: Cliff; Author
Subject: Re: [synthetos/TinyG] Turn off LEDs (#224)

Black tape.

closed 2018-04-27 by aldenhart