synthetos/TinyG #225: Setting Stepper Motor Current - VREF - tinyG V8

swpaxton66, 2018-04-26:
Hello,

I reviewed the steps to set stepper motor current here:

https://github.com/synthetos/TinyG/wiki/Connecting-TinyG

I am hoping for something more scientific.  I can't seem to find a formula that tells me what reference voltage I should use on the pots to set amperage to the stepper motors.  I am trying to maximize and/or set the proper amperage to ensure I get the maximum torque for the stepper motors.  It seems that the stepper motor is overcome too easily (skips) at my current settings.  I use the tinyG V8 and have NEMA 23 2.8A stepper motors.

I don't want to "feel" my way through the adjustment as I don't want to damage the motors.

I have not adjusted the current from the factory settings.

Can you provide this information?

Thanks!

Scott

aldenhart, 2018-04-26:
There is no direct mathematical formula we have worked out as it's specific to the motor (the reactance of the coils is a factor). The friction in the mechanism also plays a factor. The motor tuning scheme here actually works quite well.
https://github.com/synthetos/TinyG/wiki/Connecting-TinyG#setting-motor-current

With 2.8 amp steppers I would not worry about damaging the motors due to overcurrent, as that's above what the TI drivers can source. It is possible to burn out very small motors due to overcurrent - I have carelessly blown a number of NEMA 8's I use for testing, but I have never managed to blow a NEMA 17 or 23. 

That said, it is possible to source too much current and saturate the coils. If you do that the motor will run rough and hot. That's what the tuning procedure in the wiki is designed to detect and prevent.

swpaxton66, 2018-04-26:
Thanks appreciate the quick feedback.  (I have no idea what I am talking about here haha) I see your last comment about overheating, and a totally naive question, however is it possible then based on my setup and environment that I could max out the current and not have an issue?  I realize you know nothing about my setup and maybe it's an unfair question,  just exploring possibilities.

I will follow the guide you suggested.

I just did a quick check on the main tinyG site and it appears so however I would like to ask quickly.  Is V8 still the current version?

aldenhart, 2018-04-27:
It's quite possible with a 2.8A stepper you could max it out and not even get full rated torque from the motor. After a certain point, external drivers are required to get more current. Are you using these monsters here?
https://www.automationtechnologiesinc.com/products-page/stepper-motors/kl23h286-20-8b/

I have these on a Probotix and they are amazing. But I don't get the full power output they are capable of. Still, it's plenty for that application.

Yes, v8 is the current version. We are working on a completely new architecture in the g2 project but that's a lot more experimental for now.

swpaxton66, 2018-04-27:
Thanks.  I'll give it a go today.  Yes that looks like the stepper motors that I have.  Have a good one.

swpaxton66, 2018-05-04:
Hi Alden,

I would like to ask one more question here if that's ok.  I am trying to implement feed rate override.  I read the article here: 

https://github.com/synthetos/g2/wiki/Feed-Rate-Override

I tried to load the edge build but it doesn't play nice with my board.  Is it possible with my tinyG V8 to implement feed rate override?  I know John seems to control feed rate override with his Serial Port JSON Server when connecting with chilipeppr and not tinyG itself.  I use chilipeppr and CNCJS.  I checked with Cheton about his CNCJS functionality for feed rate override.  CNCJS automatically detects to see if the firmware is feed rate override capable.  He seems to have a tinyG board that I don't recognize and his version can control feed rate override.  He sent me his version:

> {fb:n}
{"r":{"fb":100.19},"f":[1,0,8]}
> {fv:n}
{"r":{"fv":0.99},"f":[1,0,8]}

Now all that said, is it possible to configure feed rate override on my specific board?

Thanks,

Scott

aldenhart, 2018-05-04:
I believe he is using the g2 code base, not TinyG. For TinyG it's best to use the Chilipeppr solution.

closed 2018-05-04 by aldenhart