synthetos/TinyG #232: TinyG stopped responding to motor movement commands

yeldarb07, 2018-07-06:
My TinG V8 stopped responding to motor movement commands. After several resets and power cycles the problem persists. I am new to programming and I'm looking for help, trouble shooting and fixing my problem. 

when I connect using CoolTerm and send a motor movement command "G0X1" the below two errors appear several times

{"er":{"fb":440.20,"st":31,"msg":"Move time is NAN"}}

followed by:

{"er":{"fb":440.20,"st":28,"msg":"Failed to get planner buffer"}}

Prior to this issue everything was working just fine. I was going through tutorials getting used to controlling and programming my XYZ axis. The last thing I did was unknowingly typed the homing command without having any limit switches installed. I either hit the reset button or turned of the power to end the command. Ever since then I can not get the TinyG to drive the motors, axis locations change but the motors never move.

yeldarb07, 2018-07-06:
I ran the homing sequence again and the Z motor responds, so I'm stuck in the homing command...

is there a way to reset this so I have control of my motors? 

Or get my limit switches wired in so I can complete the homing process?

cmcgrath5035, 2018-07-27:
Perhaps your first event corrupted some EEPROM parameters.
Run a $$ and save the results to a file , i.e. make a backup copy of your parameters.
Then execute a $defa=1 from the CLI, this will restore the parameters to the default set,
most likely not what are correct for you machine. Then reenter the parameters, from the CLI, one at a time. If you have no switches, then be sure to set $xsn=0, $xsx=0, etc. for all three axes.

Try a simple G0X1Y1 to see if you get sane movement.

If you have no limit/homing switches, what are you expecting when running homing sequence?

ril3y, 2021-04-10:
Closing due to inactivity.

closed 2021-04-10 by ril3y