synthetos/TinyG #233: Proper way to halt movement

lpd-patrick, 2018-08-13:
I am integrating a pressure sensor into the TinyG system.

What I want to do is descend to a -Z position until the sensor trips, and then halt movement or possibly return to Z0.

To detect the 'stop' point, I added the additional check to the end of the switch_rtc_callback, and it only operates while moving in the -Z direction.

What I need help with is what is the correct/proper way to halt current movement and make sure the planning system is in sync with the current location?

Then if I want to emulate returning to Z0, how do I programatically inject that into the system?

aldenhart, 2018-08-13:
You can use a G38.2 probe move. This will move in the intended axis until the probe switch closes, then stops. The position can then be read.

lpd-patrick, 2018-08-13:
Unfortunately, our sensor is not wired in as a limit switch. It's a sensor that we have to check the status of via I2C.

aldenhart, 2018-08-13:
I see. Then you probably want the sensor to initiate a feedhold. Check how the controller handles a ! character.

lpd-patrick, 2018-08-13:
What I have now is the following sequence. How does this look? Anything I missed that I should be doing?

- during specific movement, check sensor for trip point, then...
- call cm_request_feedhold(), then..
- once in hold call cm_request_queue_flush(), then...
- insert a return to 0 by calling gc_gcode_parser and reset my sequence

aldenhart, 2018-08-13:
That looks good. Try it out. Do not send any new moves once the Z move has been sent or it will flush as well.

lpd-patrick, 2018-08-15:
Preliminarily, this seems to be working.

closed 2018-08-15 by lpd-patrick