synthetos/TinyG #239: Homing of A axis stops when touching limits.

jamador47, 2018-10-23:
Hello! I have 2 Y axis on my machine, so I intend to control them independently through the Y and A axis. I have configured all the axes and they are working fine except for AXIS A, as soon as it touches the homing limit it stops and does not continue to do the zero backoff procedure. I have verified that the limit is configured as 1 (homing only). 

When this happens, the machine does not enter an alarm state, and it allows me to do other operations. I have also tried disabling the softswitches and the problem persists. Any ideas?

troisiemetype, 2018-10-24:
Hello !
Does your machine really have two Y axis ? Or does it have one Y axis with two motors ?
The problem is, A is a rotary axis, I believe it's not handle the same way as X, Y and  Z.

ril3y, 2026-11-16:
Please retest on 450.06 with `$aam=1` and post your `$a` settings if A still stops on the switch without backing off. Several homing fixes since 450.01 may cover this.

closed 2026-11-16 by ril3y