synthetos/TinyG #244: TinyGv8 serial not being recognized

vfrman1, 2019-03-26:
I'm having an issue with my tinyg for my shapeoko2.  I was vacuuming off the waste board between jobs and the CNC was sitting idle, and powered on. I got a pretty good shock when I touched the gantry and now the  tinyg won't talk to the pc.  When I plug in the board, the spdir light flashes for 3-4 seconds, then goes out.  Other lights on are the power, tx, and PWM.  I have a 24v fan that also comes on.  The board is getting power, but it isn't communicating with the PC.

I'm running win10 on a 64 bit computer.  When the board is powered up, there is no sound from the computer recognizing the usb and when I run the JSON server, it returns a "there are no serial ports to list".  When I check the device manager in windows, COM3 is gone, which is what the tinyg used to connect to.  

I've tried a reset to the board, rebooting the computer, and changing out the usb cable.  All these actions did nothing.

cmcgrath5035, 2019-03-31:
From your description,tinyG is booting up normally.
A good guess would be that the ESD event you describe ("pretty good shock") has damaged or destroyed the FTDI device (USB to Parallel converter).

If your computer cannot see the USB port to set it up, communications won't get started.
Are you sure that the Computer's usb port is still functioning? Try to plug in a memory stick or other USB device.

High velocity shop vacs, with all plastic hoses and tools, in combination with dry wood chips or dry plastic flakes are really good ESD generators

vfrman1, 2019-03-31:
That was pretty much what happened.  I received a new board yesterday, got it all wired in and working.  Is there any way to fix/replace the FTDI device?

ril3y, 2021-04-09:
I am going to close this issue as its not really a software issue.

closed 2021-04-09 by ril3y