synthetos/TinyG #247: G3 not working

3rdScout, 2019-05-20:
G3 with I & J values only goes to start position and end position in a straight line? I've tried different post processors.

Code example below:
%
N1 G0 X1.7479 Y-0.2694 Z0.1250
N2 G1 Z-0.125015
N3 G3 X0.4966 Y14.8794 I-12.1536 J6.6222
N4 X-0.4978 Y14.8779 I-0.4966 J-0.3794
N5 X-1.7493 Y-0.2545 I10.9347 J-8.5223
N6 G2 X1.7479 Y-0.2694 I1.7453 J-0.7835
N7 G1 Z-0.2500

cmcgrath5035, 2019-07-09:
I am guessing this might be inch mode G code ?
Try regenerating in mm mode.

ril3y, 2019-07-29:
Did this work?

3rdScout, 2019-07-30:
I run Ubuntu. Sometimes, if my computer has been running for awhile with other programs, or if i started and stopped the server by closing the terminal window without killing  the process then hence the issue. A clean reboot seems to fix the issue. Hope this helps.

ril3y, 2021-04-10:
Closing due to inactivity.

closed 2021-04-10 by ril3y