synthetos/TinyG #248: Python Control

bashfulporpoise, 2019-07-23:
I am trying to send simple one line commands to TinyG using a python script I am writing. Using PySerial, I can connect and recieve data from the TInyG board but when I try to send data back to the board, I get no response and no action performed. The Rx light flashes when I try to send data using my script but nothing comes of it. I have tried sending my code as json (Ex: {"x":n}, {"x":"n"}, and even $1pm=0) I have also tried to run plain text in the form of gcode commands but I get nothing. Everything being sent is using python's .encode() utility for proper transmission. I suspect there may be some intial parameters I have not set up but there is no mention of them in the wiki that I have found.

siberianfox, 2019-07-30:
I think you can use a serial port monitor to locate the problem. You said that the RX light flash when you try to send data, maybe the baud rate is no correct?

ril3y, 2019-07-30:
make sure you are sending a \n at the end of the command.

bashfulporpoise, 2019-07-31:
The baudrate is correct but I am not sure if the \n was sent I will try that now

bashfulporpoise, 2019-07-31:
I tried with the \n with the command but it did not work. Is there a way to use the g2core api in python or would I have to code it in Javascript? I don't have any experience in javascript and my python experience is limited as well.

bashfulporpoise, 2019-07-31:
I believe I am running into problems with this section. "The host sends a few command lines to prime the board's serial receive queue, then the host sends a new line each time it receives a response from a processed line. That's it. So if you are building a sender that's all you need to do."

houda84, 2020-04-19:
Did you encoded it in bytes ? ex :  serial.Serial.write(bytes('{"x":n}', 'ascii'))

ril3y, 2021-04-09:
closing due to lack of activity

closed 2021-04-09 by ril3y