synthetos/TinyG #249: Linux toolchain?

norru, 2019-07-28:
Hello,

I am evaluating TinyG as a possible option for a semi-custom CNC build.

A must-have feature to me is the ability to build and upgrade the firmware from a Linux host. I do not need it to be easy, just make sure it's possible.

- Is the avr-gcc + avrdude toolchain enough?
- Is Atmel Studio (no Linux availability) required at any stage?
- Can the upgrade from binaries be done without your proprietary app (which also doesn't appear to run on Linux)?

Many thanks.

DanalEstes, 2019-07-28:
I am not the author, but I have worked with TinyG and the firmware quite a
bit.

Third question first:  Loading a board to install new binaries can be done
with any tool that will serial load a "bootloader".  avrdude, etc.  This
means you can experiment with the last few releases and find which one
works best for your applications.

For the first two questions: You should be aware that this firmware is no
longer being developed actively by synthetos.  There has not been a commit
(nor release) that actually changed anything in four or five years.  Unless
you intend to make major changes, there is no need to compile from source,
ever.

Having said that, I still run TinyG boards in several machines.  The
smoothness of the "constant jerk" motion planner just can't be beat.  Is
is/was very disappointing to me to see development stop.

Danal

norru, 2019-07-29:
> Is is/was very disappointing to me to see development stop.

Thanks for the very helpful comment. Just for this reason alone I'd rather look elsewhere :)

Many thanks!

closed 2019-07-29 by norru

ril3y, 2019-07-29:
TinyG is no longer being developed but G2core is.  Just an fyi.

Riley

norru, 2019-07-29:
@ril3y doesn't G2Core run on TinyG2 boards? I could not find any of those.

ril3y, 2019-07-29:
You can run g2core on the arduino due with a gShield from our site for 3
axis setups.  We are still working out a few hardware issues but hopefully
should release our own g2core boards soon.

norru, 2019-07-29:
Never mind, I have ordered an Arduino Due clone to tinker with g2core :). I'll keep you posted.

norru, 2019-07-30:
Hi, I've seen the g2core boards but I need a XYYZ configuration which requires 4 drivers.

norru, 2019-08-01:
@ril3y Hi, I think I will try to build my own "gShield" as I want to control 4 TB6600 standalone drivers off-board. I still would like to use the gShield itself as the building base. Do you have kicad schematics I can nick :) ?

ril3y, 2019-08-01:
There is this.

https://github.com/synthetos/grblShield/tree/master/hardware

I think that is all we provide.

norru, 2019-08-02:
Looks good enough for what I need :) what's the IC part?