synthetos/TinyG #250: Zmin Probe G38.2 Not Working

swpaxton66, 2019-08-03:
Hello!

I have a V0.97 TinyG with firmware build 440.20.

I had a touch plate Z probe setup working.  Now it doesn't work.

I've confirmed no broken wires by continuity test.

I've set zsn to all possible settings.

Nothing fancy just a roach clip and a touch pad.  When there is continuity it should trigger the tinyG to set Z zero.  It just keeps on chugging.

Anyone have any ideas?

I'm thinking a board issue as it was workign in the past.  As far as I know I didn't change anything to affect it.

Thanks.

swpaxton66, 2019-08-03:
Forgot to mention.

st = 0

cmcgrath5035, 2020-04-15:
G38.2 is used for auto leveling. Should you be using G28.2 to Home Z using your touch plate?

I am not clear what "It just keeps on chugging" means

swpaxton66, 2020-04-15:
Hi thanks for the feedback.  That was a typo on the GCode command.  "keeps chugging" = means that once it contacts the touch plate it just keeps moving the z-axis down and doesn't register the hit.

cmcgrath5035, 2020-04-15:
Give https://github.com/synthetos/TinyG/wiki/Homing-and-Limits-Description-and-Operation a read.
Do you have Z min set as home as well as NO (st=0)?
Otherwise, it sounds like you are using this process https://github.com/synthetos/TinyG/wiki/Automatic-Z-Homing-When-Milling-PCBs
Here is a thread using G38.2 probing https://openbuilds.com/threads/touch-plate-sends-z-axis-in-the-wrong-direction.11643/

ril3y, 2026-11-16:
Closing as stale. On 450.06, set the probe input with `$zsn=4` and its polarity with `$prbp`, then check `{in:n}` while touching the plate; please reopen with that output if it still does not trip.

closed 2026-11-16 by ril3y