synthetos/TinyG #254: PWM not working

bionicbeaver1, 2020-07-31:
Do you need to connect a laser driver to the pwm to run the laser or it will drive it directly. cant find the info anywhere.

nikolac, 2020-09-22:
You have any documentation on your laser driver? I'm no expert, but I had issues with PWM and my spindle driver. Specifically, the default PWM frequency was mismatched.  

[p1frq] pwm frequency 15000 Hz

bionicbeaver1, 2020-09-22:
I thought I had my laser info on here but I guess not, I have a 2.5 laser that I just got working yesterday, I am using a tinyG control and a pwm motor driver to run the laser, I guess if you only have the laser module then you need to have a laser driver.
Sorry not very good at this, only took five years to get it running still not sure its correct.

ril3y, 2021-04-10:
Closing due to inactivity.

closed 2021-04-10 by ril3y