synthetos/TinyG #263: Unable to clone repo on windows. Repo contains a file named NUL

VillageID10T, 2021-04-09:
Path: firmware/tinyg/default/NUL

git on windows is unable to clone this file.

ril3y, 2021-04-09:
This works just fine for me.

`PS D:\GitHub> git clone https://github.com/synthetos/TinyG.git ddd
Cloning into 'ddd'...
remote: Enumerating objects: 71, done.
remote: Counting objects: 100% (71/71), done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 32470 (delta 40), reused 38 (delta 19), pack-reused 32399
Receiving objects: 100% (32470/32470), 161.24 MiB | 10.08 MiB/s, done.
Resolving deltas: 100% (24142/24142), done.`

closed 2021-04-09 by ril3y

VillageID10T, 2021-04-09:
awesome thanks! looks like it was fixed in a PR earlier today.

ril3y, 2021-04-10:
@VillageID10T yah bud np.  We have been SUPER BEHIND on any support for TinyG since we are focused on G2core 100%.  However, just doing some maintance.  We accept pull requests for sure!