synthetos/TinyG #264: Error in source code

DmitriiZorin, 2021-04-29:
https://github.com/synthetos/TinyG/blob/39df38c0e5830721fb2d16a7ae7b44816d24c4c4/firmware/tinyg/plan_exec.c#L382
Hello. I think that the line 382 in file plan_exec.c meant to be: 
float half_Ah_5 = A * half_h * half_h * half_h * half_h * half_h;
Am i right or did i misunderstood something?

lagnat, 2021-04-29:
You're probably right and it looks like that bug was already fixed in G2.

https://github.com/synthetos/g2/blob/edge/g2core/plan_exec.cpp#L728

ril3y, 2026-11-16:
Fixed: the first-segment velocity now uses the A coefficient as you described (`half_Ah_5 = A * ...` in `plan_exec.c`), in every 450.x release.

A host test runs the shipped `_init_forward_diffs()` and checks the first segment and every later segment midpoint against the closed-form velocity profile; with the old C coefficient restored it fails (750 vs 900 mm/min for a one-segment 1500 -> 300 mm/min deceleration).

Test: [tests/host/test_e014_first_segment_velocity.py](https://github.com/synthetos/TinyG/blob/03cc86e5/tests/host/test_e014_first_segment_velocity.py) (edge-next 03cc86e5).

closed 2026-11-16 by ril3y