synthetos/TinyG #266: M0 is ignored?

jrcandy, 2021-06-07:
Hiya,
[Gcode support](https://github.com/synthetos/TinyG/wiki/Gcode-Support) says `M0` is supported but when I have it in the middle of a program it is ignored. I have sent this program via UGS and also chilipeppr with the same result.

The program can be as simple as
`G90 G0 X0
M0
G0 X-5`

But I wonder how the resume signal would be sent anyway since there's no physical 'resume' button in TinyG board ? It's all good in UGS / chili with their UI, but we'll be sending gcode programatically, so I'm confused how it would resume as there isn't a M-command for 'resume'

Replacing an M0 for a `!` in the gcode doesn't work as expected because as soon as it's read that chunk of gcode in the buffer it will stop, not when I want it to stop. Technically M0 is not the same as Feed hold, isn't it?

thanks for your efforts!

Milner99, 2021-06-07:
Experiencing the same.

ril3y, 2026-11-16:
Fixed in 450.01: M0 now pauses the program in queue order, and `~` (cycle start) resumes it, also when sent programmatically. Unlike `!`, which stops at once, M0 stops exactly where it sits in the program.

Fix: https://github.com/synthetos/TinyG/commit/f3418d2e (first released in [450.01](https://github.com/synthetos/TinyG/releases/tag/450.01))
Regression test: `tests/hil/tests/bugs/test_e020_m0_does_not_pause.py`

closed 2026-11-16 by ril3y