synthetos/TinyG #268: M115 code unsupported

fxrb, 2021-06-25:
A TinyG board with firmware build 440.20 and firmware version 0.97 reportes `err: M code unsupported: M115` when I send this code using a terminal. Same for M114.
These M codes are supposed to be supported since version 0.95, correct?

kierzie, 2022-12-02:
I also have a TinyG board with firmware build 440.20 and firmware version 0.97 that reports M code unsupported for M115 and other M codes I was counting on using. I used CoolTerm:
tinyg [inch] err: M code unsupported: M115 
tinyg [inch] err: M code unsupported: M114 
tinyg [inch] err: M code unsupported: M400

ril3y, 2026-11-16:
M114, M115 and M400 are supported in 450.01 and later (edge-next releases). 440.20 did not include them. See the Gcode Support wiki page for the reply formats.

Fix: https://github.com/synthetos/TinyG/commit/99524cc0, https://github.com/synthetos/TinyG/commit/03d3bef3 (first released in [450.01](https://github.com/synthetos/TinyG/releases/tag/450.01))
Regression test: `tests/hil/tests/bugs/test_e001_m114_m115.py`, `tests/hil/tests/bugs/test_e002_m400_wait.py`

closed 2026-11-16 by ril3y