synthetos/TinyG #270: G28.2 move starts, stops, and then continues

natd89, 2021-07-23:
When I run the homing command the axis moves ~5cm pauses and then resumes the motion. If I click the limit switch between when it first starts moving and initially pauses, the motor stops turning and then resumes motion before I let go of the limit switch. When I let go of the limit switch the motor stops and the board is not homed when I check $home.

If I wait long enough after the G28.2 command is sent, the motor will start running, pause for a little after moving a short distance and then resume motion. If I press the limit switch after it resumes motion the axis homes correctly. 

My concern with this is that if I try homing the machine with the carriage too close to the limit switch the machine will crash because there isn't enough space for the axis to start moving, pause, and then resume/start the homing sequence. Any idea why this is happening?

duchonic, 2023-07-04:
We use the TinyG for the Liteplacer.com project. We had the same effect and i think it's because the Max Switch is On.
If the Max Switch is On the Axis will move out of the Max Switch before homing with the Min Switch.

Our problem was that the Max Switch was not wired correctly (wrong logic with normaly closed).
You can also try to disable the Max Switch for homing.

ril3y, 2026-11-16:
The short first move is the homing clear move, run when a switch reads active at the start of the cycle. Two fixes cover the behaviour described here: with no limit switch at the other end the clear move read an invalid switch index and could move unpredictably (fixed in 450.03), and a switch still active after the clear move (for example a mis-wired NC/NO switch, as discussed in the comments) now fails homing instead of homing anywhere (fixed in 450.01).

Tests: [tests/host/test_n73_homing_limit_index.py](https://github.com/synthetos/TinyG/blob/03cc86e5/tests/host/test_n73_homing_limit_index.py) and [tests/hil/tests/bugs/test_e132_homing_stale_switch.py](https://github.com/synthetos/TinyG/blob/03cc86e5/tests/hil/tests/bugs/test_e132_homing_stale_switch.py) (edge-next 03cc86e5). Please reopen if you see it on 450.06.

closed 2026-11-16 by ril3y