synthetos/TinyG #273: feedrate override with cncjs

Kbex1, 2021-09-24:
i think i might be missing something here,
i'm running the latest firmware on a tinyG v8 (Edge-449.01) and the lastest cncjs 1.9.22.
with JSON enabled {sso:n} is not recognized, and is not mentioned in the docs.
it is not supported anymore ?  (or the more likely scenario i'm missing some configuration )

Thanks

kbex1, 2021-09-26:
after some digging  i found the the spindle code that supports the feedrate is only on branch edge 0.98 and was never merged into the other "edge" branch master-449.

@aldenhart is there something i should be aware of before i merge the 0.98 spindle code that supports the federate override to the master-449 branch ? 

i can test it with my machine but before i solve the merge conflicts for the "master-449" branch and such, is there a reason just taking the code from edge-0.98 for the spindle.h and spindle.c and compiling shouldn't work? it seems there was only one commit for it in 444.18 on the edge-0.98 branch.

thanks
kfir

ril3y, 2026-11-15:
Feed rate and spindle speed override are available in firmware build 450.06 (edge-next pre-release): https://github.com/synthetos/TinyG/releases/tag/450.06

They use the same names as g2core, so a host that already drives g2core's overrides can send the same commands:

- `{"mfo":0.5}` (or `$mfo=0.5`) runs the job at 50% feed, anywhere from 5% to 200%. It applies to every feed move (`G1`, `G2`, `G3`), including moves already queued, and blends in within the machine's acceleration and jerk limits.
- `{"sso":0.5}` (or `$sso=0.5`) runs the spindle at 50% of the programmed `S`, from 5% to 200%, limited to the configured spindle speed range. It never starts a stopped spindle.
- Both are on by default (`mfoe`, `ssoe` and `m48e` are 1) and are never saved to EEPROM, so they are accepted while a job runs. The program can turn them off with `M49`, `M50 P0` (feed) or `M51 P0` (spindle). Reset, ^D and `M2`/`M30` return both to 100%.

Details: the Feed Override wiki page, https://github.com/synthetos/TinyG/wiki/Feed-Override

Build 450.05 had a first version of feed override that only slowed long straight moves; use 450.06 or later. Whether a button in your sender (for example CNCjs) drives these depends on the sender version.

closed 2026-11-15 by ril3y