synthetos/TinyG #282: Using TinyG in WPF Project

David-Vice, 2024-01-05:
Hello, I have a big project where the motors are used to move the arms of goniospectrophotometer. And as I understood WPF causes some thread related problems, because TinyG driver, which is in C++ is not thread safe. I put mutexes in the source code of C++ and WPF project and now it works better, but still not ideal. If I put getranges() 10 times in a row, on 3rd or 4th try it will give some corrupted values like 1.323411414 or 2.12312441. Who knows how to fix it, or maybe someone has thread safe version of this driver.

Thank you in advance!

ril3y, 2024-01-05:
Tinyg uses a serial port to send data.  It also maintains a state in the
firmware.  If you are writing to the serial port with multiple threads,
while this could be possible I think your design is flawed.   You should
setup a thread to write data to the board and then use a queue etc to put
data that needs to be sent.  You could then have threads push data into
that queue.  Not understanding what you are going for really.  This is more
of a general software design follow up.

Riley

David-Vice, 2024-01-07:
Thank you, but I'm talking about cpp version of TinyG driver, that was sent to me. I created C++/CLI wrapper for it and added mutexes to make it thread safe, but it doesn't work perfect still. Maybe you have an idea how I can fix it.

Repo: https://github.com/David-Vice/Thread-Safe-Wrapper

Thank you in advance!

siredmar, 2024-01-07:
Just out of curiosity... What DLL are you referring to?

David-Vice, 2024-01-07:
> Just out of curiosity... What DLL are you referring to?

The code I gave generates Optel_TinyG.dll and isomove.dll

ril3y, 2024-01-07:
This is not our code. You should talk to the author of that code.

riley

siredmar, 2024-01-08:
Not sure who or what Optel is, but i second @ril3y. Unless you are not the author, you should search out for help where you found that source.

ril3y, 2026-11-16:
This is about a third-party driver, so please contact its author. TinyG itself needs a single writer thread with a queue on the host side.

closed 2026-11-16 by ril3y