synthetos/TinyG #283: TinyG CNC won't cut arrayed circle/archs: TinyG post processor in UGS

Midlyhelpless, 2024-05-20:
### My CNC uses: 
TinyG v8 board
files made in Vcarve Pro 11.5
TinyG inches as post processor
UGS as the interface
ASUS laptop as controller

When trying to place dogholes on the spoilboard it keeps pausing at the same spot in the arch. but it doesn't do it for every doghole or all the depth passes. It also doesn't want to start the file at all except for! It would allow me to resume if I spammed the play button (it would often re-pause the file after the first and second press) 

So to get the file started I have to mash the play button and at certain similar points it will pause again and I'll re-mash the play button to keep the file going ... after three of four rounds of button mashing UGS will throw a Unhandled message when no limits have been triggered. UGS also doesn't show any error code in console and everything mentioned also happens when cutting air.

An import detail is that the machine WILL flawlessly cut one single test doghole file of the exact same size, speed, location as the full file but the file does not contain any extra doghole data. The Gcode files differs in some notable ways but both files are made in Vcarve almost exactly the same the only difference is instead of: making one circle - saving toolpath. I make one circle - array the circle - array the toolpath - then save toolpath (this array action might be a problem ill test it out)

[image]
[image]
[image]

https://github.com/synthetos/TinyG/assets/170275392/f666d141-f13a-477b-9343-eac60547a1da

Midlyhelpless, 2024-05-21:
The issue has been solved. The PostProcessor needed to be TinyG(mm) not TinyG(In) regardless of what the UGS Jog interface says. You can certainly change the Jog interface to metric or imperial based on preference but either/or should work even though the PostProcessor is metric.

closed 2024-05-21 by Midlyhelpless

ril3y, 2024-05-21:
Thanks for updating this!